aerostack2/aerostack2 #740: [as2_map_server] Add basic gtest to check plugin loading

pariaspe, 2025-01-14:
You should add a basic gtest that load the plugin to verify that it works, as the rest of aerostack2 packages

_Originally posted by @RPS98 in https://github.com/aerostack2/aerostack2/pull/737#pullrequestreview-2536684078_

hmmnice, 2026-11-05:
Hi, I’d like to work on this. I noticed the existing constructor test already attempts to load the default scan plugin. I’m proposing a focused gtest that directly checks discovery and instance creation of scan2occ_grid::Plugin through pluginlib. Since the depth and point-cloud launchers reuse that plugin, would this satisfy the intended scope, or were you looking for launch-level coverage too?

pariaspe, 2026-11-09:
Hi, plugin instantiation gtest would be enough. Feel free to open a PR, no assignation is needed.

closed 2026-11-11 by pariaspe