aerostack2/aerostack2 #749: Build issue when compiling project using clang

saitejakuchi, 2025-02-13:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

Hello team, 

I am trying to build the project using clang instead of gcc and encountering the following ABI issue.

/usr/bin/ld: /ros2_ws/install/as2_core/lib/libmocks.so: undefined reference to `as2::control_mode::controlModeToString[abi:cxx11](as2_msgs::msg::ControlMode_<std::allocator<void> > const&)'
clang: error: linker command failed with exit code 1

I tried compiling with CMAKE_CXX_STANDARD set to 11 and 14. added few extra flags "-stdlib=libstdc++, -D_GLIBCXX_USE_CXX11_ABI=1". But nothing helped. Is there a way to compile the project using clang-14?

### How to reproduce your issue?

Compiling the project using clang-14.

### Relevant log output

```shell

```

miferco97, 2025-02-13:
Hello! this might be an issue with clang standard libraries.
As seen in this [stack overflow post](https://stackoverflow.com/questions/74785927/clangd-doesnt-recognize-standard-headers)
The solution might be running:
```
sudo apt install libstdc++-12-dev
````
Let me know if this helps :)

saitejakuchi, 2025-02-13:
Nope, that didn't help. Are you able to build using clang?

miferco97, 2025-02-14:
I think I have found the issue, and should be solved in the #750 Pull request.
Could you check if it works on this branch before merging ?

closed 2025-02-14 by RPS98

saitejakuchi, 2025-02-16:
Yup, that works. Thank you.