aerostack2/aerostack2 #768: AS2 message not found

migueltg20, 2025-03-15:
Hi, 
I am trying to figure out how to use the PID controller plugin and I have seen that a message called "trayectory_setpoints" is used. My problem is that I do not find this message in the files of my binary installation. Is it "used" from a CMakeLists like the package "pid_controller" or do I have to install something additional?

Thanks.

pariaspe, 2025-03-16:
Hi @migueltg20 
Try to list your ros2 interfaces and check that `as2_msgs/msg/TrajectorySetpoints` is in the list.
```bash
ros2 interface list -m
....
    as2_msgs/msg/Acro
    as2_msgs/msg/AlertEvent
    as2_msgs/msg/BehaviorStatus
    as2_msgs/msg/ControlMode
    as2_msgs/msg/ControllerInfo
    as2_msgs/msg/FollowTargetInfo
    as2_msgs/msg/Geozone
    as2_msgs/msg/GimbalControl
    as2_msgs/msg/MissionEvent
    as2_msgs/msg/MissionUpdate
    as2_msgs/msg/NodeStatus
    as2_msgs/msg/PlatformInfo
    as2_msgs/msg/PlatformStateMachineEvent
    as2_msgs/msg/PlatformStatus
    as2_msgs/msg/PolygonList
    as2_msgs/msg/PoseStampedWithID
    as2_msgs/msg/PoseStampedWithIDArray
    as2_msgs/msg/PoseWithID
    as2_msgs/msg/Speed
    as2_msgs/msg/Thrust
    as2_msgs/msg/TrajGenInfo
    as2_msgs/msg/TrajectoryPoint
    as2_msgs/msg/TrajectorySetpoints
    as2_msgs/msg/YawMode
....
```

You should have it installed in `as2_msgs` package.

migueltg20, 2025-03-17:
If I do that, the only message interface that I don't have is ```as2_msgs/msg/TrajectorySetpoints```:
```
ros2 interface list -m
. . .
    as2_msgs/msg/AlertEvent
    as2_msgs/msg/BehaviorStatus
    as2_msgs/msg/ControlMode
    as2_msgs/msg/ControllerInfo
    as2_msgs/msg/FollowTargetInfo
    as2_msgs/msg/Geozone
    as2_msgs/msg/GimbalControl
    as2_msgs/msg/MissionEvent
    as2_msgs/msg/MissionUpdate
    as2_msgs/msg/NodeStatus
    as2_msgs/msg/PlatformInfo
    as2_msgs/msg/PlatformStateMachineEvent
    as2_msgs/msg/PlatformStatus
    as2_msgs/msg/PolygonList
    as2_msgs/msg/PoseStampedWithID
    as2_msgs/msg/PoseWithID
    as2_msgs/msg/Speed
    as2_msgs/msg/Thrust
    as2_msgs/msg/TrajGenInfo
    as2_msgs/msg/TrajectoryPoint
    as2_msgs/msg/YawMode
. . .
```

RPS98, 2025-03-17:
Hi @migueltg20 

The message has changed since the last release. Could you specify what type of installation you have?

In the binary version, `TrajectorySetpoint` is used, but in main, `TrajectorySetpoints` is currently used.

If you are using the source version, I recommend cleaning the workspace and compiling it with the latest main version.

migueltg20, 2025-03-18:
I have the binary version. I have tried to uninstall the package and install it back but the message is still lost (with and without the "s" at the end of the name). 

Moreover, if I try to reinstall the whole AS2 framework (binary version), the same thing happens.

Should I try the source installation?

RPS98, 2025-03-18:
Hi @migueltg20,

It's strange, with everything in binaries, it should work, as TrajectoryPoint.msg is available: https://github.com/aerostack2/aerostack2/blob/1.1.2/as2_msgs/msg/TrajectoryPoint.msg

You can also try using Docker.

The choice between using sources or binaries depends on the application you want to use Aerostack2 for.

From source, you can see all msgs available: https://github.com/aerostack2/aerostack2/tree/main/as2_msgs/msg

pariaspe, 2025-03-18:
> Should I try the source installation?

If you don't mind, give it a try. We usually work with the source version, so we will probably be more helpful.

closed 2025-03-22 by migueltg20